johncclayton/electric #3: Read/write failures on the USB bus should be retried automatically by the server code.

johncclayton, 2017-01-02:
Sometimes reads/writes via the USB-HID interface don't work, and simply retrying solves the problem.  The server should retry up to 5 times with a half second pause between each request. 

During retries, the server should also attempt to find the device again - if the device is NOT found then the request should no longer be retried, instead it should be stopped and a "disconnected charger" state is returned for the request.

scornflake, 2017-01-02:
not necessarily?  This can be done really easily on the rxjs side? so assuming you return an error response....  

or use a reactivex impl on server side as well

johncclayton, 2017-01-03:
ok, so I'll close this and you just yell if you need it

closed 2017-01-03 by johncclayton