johncclayton/electric #13: I think the server may be barfing after quite some time

scornflake, 2017-01-05:
Exception AttributeError: "'iChargerMaster' object has no attribute '_is_opened'" in <bound method iChargerMaster.__del__ of <electric.icharger.modbus_usb.iChargerMaster object at 0x6cdcabd0>> ignored

It seems to stop the server

johncclayton, 2017-01-05:
Need to get more details as to the context, I've seen this to and this
error is *definately* a problem with modbus_tk using the .close() method
within the __del__ - whereby apparently there are no guarantees that the
instance vars are still around within __del__, so bad stuff happens - e.g.
NOT best practise.

Thanks,
--
John Clayton
Skype: johncclayton

johncclayton, 2017-07-22:
Again, based on todays work/testing I think this issue is no longer relevant.

closed 2017-07-22 by johncclayton