johncclayton/electric #16: With no pack plugged in, get a weird high voltage

scornflake, 2017-01-05:
"curr_out_volts": 33.264

This is on a 308. John mentioned that he saw +40 (ish) for his 4010.
Maybe the protocol says "if I don't know the voltage, I'll give you something > max input volts".

This would mean we'd need to have a "Charger" class on the server, that knows something about each model (i.e: max input voltage).

Ideally, the client wants to see 0.0 for curr_out_volts when no pack is plugged in (main charge lead).

johncclayton, 2017-01-06:
Checking if balance leads are plugged in is easy.  Knowing whether the batt is plugged in is tricky as the output voltage spikes to the following: number of physical charger channels * 4.2.

The calculation is only rough - but its within 0.5v accurate.   So, for a 4010 with a total of 8

scornflake, 2017-01-13:
blah blah blah.

can you please set 'curr_out_volts' to 0 when balance_leads_plugged_in is false and battery_plugged_in is false.

scornflake, 2017-01-14:
I've put something in place, and it's truly, truly a thing of sick.

Think: globals
So, it's not going to work in production. The prob is that I need the data in ChannelStatus, but what I need comes from /Status (DeviceInfo).

I don't really want channel to do a DeviceInfo request... but given the stateless nature of the server, where do I put the last seen value?

yik.

closed 2017-01-14 by scornflake