johncclayton/electric #25: Lets talk nginx for hosting (flask isn't that great)

scornflake, 2017-01-08:
I get LOTS of pauses when testing, and often have to restart the server.

I "presume" this to be flask? It certainly doesn't handle more than a few req's p/s, and if I want to start upping the poll frequency on the client, I guess it might be a problem.

Do we intend to ship with a proper hosting front end?

johncclayton, 2017-01-08:
How about we use something that's Python based like Gunicorn, Twisted, GEvent etc - better than the built in Flask one and yet no additional behemoth to deploy and configure.  If you want nginx then I'd start thinking harder about deploying using Docker - which suddenly means we can't just "run it on windows" or "rasp pi"

johncclayton, 2017-01-08:
I'm now using gunicorn - it can scale out, and out of the box should handle easily a few hundred requests per second.  I won't close this until I've updated the PyPi electric-server scripts to use it as well.

johncclayton, 2017-01-19:
Closing out - gunicorn + evil_globals is doing ok.

closed 2017-01-19 by johncclayton