johncclayton/electric #30: Is cell_total_voltage accurate?

scornflake, 2017-01-09:
I see a value of 18.421, but this doesn't show in the charger anywhere.

I'm recalling looking to see if the charger itself provides the Sv (cell voltage sum) value (16.373 in the pic). No biggie if it doesn't. But I did notice this "18" value and thought "BETTER REPORT THAT TO THE SERVER SIDE TEAM"

I've a 4S plugged in (see pic).
[image]

scornflake, 2017-01-09:
{
    "balance_leads_plugged_in": true, 
    "battery_plugged_in": true, 
    "cell_count_with_voltage_values": 6, 
    "cell_total_ir": 0.0, 
    "cell_total_voltage": 18.416000000000004, 
    "cells": [
        {
            "balance": 0, 
            "cell": 0, 
            "ir": 0.0, 
            "v": 4.089
        }, 
        {
            "balance": 0, 
            "cell": 1, 
            "ir": 0.0, 
            "v": 4.093
        }, 
        {
            "balance": 0, 
            "cell": 2, 
            "ir": 0.0, 
            "v": 4.092
        }, 
        {
            "balance": 0, 
            "cell": 3, 
            "ir": 0.0, 
            "v": 4.094
        }, 
        {
            "balance": 0, 
            "cell": 4, 
            "ir": 0.0, 
            "v": 0.0
        }, 
        {
            "balance": 0, 
            "cell": 5, 
            "ir": 0.0, 
            "v": 0.0
        }, 
        {
            "balance": 0, 
            "cell": 6, 
            "ir": 0.0, 
            "v": 0.0
        }, 
        {
            "balance": 0, 
            "cell": 7, 
            "ir": 0.0, 
            "v": 0.0
        }, 
        {
            "balance": 0, 
            "cell": 8, 
            "ir": 102.4, 
            "v": 1.024
        }, 
        {
            "balance": 4, 
            "cell": 9, 
            "ir": 102.4, 
            "v": 1.024
        }
    ], 
    "channel": 0, 
    "charger_presence": "connected", 
    "control_status": 0, 
    "curr_ext_temp": 25.4, 
    "curr_inp_volts": 22.839, 
    "curr_int_temp": 39.1, 
    "curr_out_amps": 0.0, 
    "curr_out_capacity": 0, 
    "curr_out_power": 0.0, 
    "curr_out_volts": 16.372, 
    "cycle_count": 0, 
    "dlg_box_id": 0, 
    "line_intern_resistance": 0.0, 
    "run_error": 0, 
    "run_status": 0, 
    "timestamp": 0.0
}

johncclayton, 2017-01-09:
You front end guys.... my goodness.

closed 2017-01-09 by johncclayton