johncclayton/electric #31: Seeing preset with empty name in the JSON

scornflake, 2017-01-09:
One has an empty name (the 2nd, in my case).  It doesn't show on the charger.

https://www.dropbox.com/s/7xfv5pct3nv5f7c/File%209-01-17%2C%2010%2017%2033%20PM.mov?dl=0

JSON:
[
    {
        "auto_save": true, 
        "bal_delay": 1, 
        "bal_diff": 5, 
        "bal_over_point": 0, 
        "bal_set_point": 5, 
        "bal_speed": 1, 
        "bal_start_mode": 2, 
        "bal_start_voltage": 3, 
        "capacity": 0, 
        "channel_mode": 0, 
        "charge_current": 1.0, 
        "cycle_count": 3, 
        "cycle_delay": 3, 
        "cycle_mode": 0, 
        "discharge_current": 5.0, 
        "end_charge": 10.0, 
        "end_discharge": 50.0, 
        "fast_store": true, 
        "index": 0, 
        "keep_charge_enable": false, 
        "li_balance_end_mode": 0, 
        "li_cell": 0, 
        "li_mode_c": 0, 
        "li_mode_d": 0, 
        "life_charge_cell_voltage": 3.6, 
        "life_discharge_cell_voltage": 2.5, 
        "life_storage_cell_voltage": 3.3, 
        "lilo_charge_cell_voltage": 4.1, 
        "lilo_discharge_cell_voltage": 3.5, 
        "lilo_storage_cell_voltage": 3.75, 
        "lipo_charge_cell_voltage": 4.2, 
        "lipo_discharge_cell_voltage": 3.5, 
        "lipo_storage_cell_voltage": 3.85, 
        "log_interval_sec": 1.0, 
        "name": "LiPo", 
        "ni_cell": 0, 
        "ni_discharge_voltage": 0.01, 
        "ni_mode_c": 0, 
        "ni_mode_d": 0, 
        "ni_peak": 3.0, 
        "ni_peak_delay": 3, 
        "ni_trickle_current": 0.05, 
        "ni_trickle_enable": false, 
        "ni_trickle_time": 5, 
        "ni_zero_enable": false, 
        "ni_zn_cell": 0, 
        "ni_zn_charge_cell_volt": 1.9, 
        "ni_zn_discharge_cell_volt": 1.1, 
        "op_enable_mask": 65535, 
        "pb_cell": 6, 
        "pb_cell_float_enable": false, 
        "pb_cell_float_voltage": 2.3, 
        "pb_charge_voltage": 2.4, 
        "pb_discharge_voltage": 1.8, 
        "pb_mode_c": 0, 
        "pb_mode_d": 0, 
        "reg_ch_current": 1.0, 
        "reg_ch_mode": 0, 
        "reg_ch_volt": 12.0, 
        "regen_discharge_mode": 0, 
        "restore_current": 0.1, 
        "restore_time": 3, 
        "restore_voltage": 1.0, 
[... 1144 lines cut ...]
        "pb_charge_voltage": 2.4, 
        "pb_discharge_voltage": 1.8, 
        "pb_mode_c": 0, 
        "pb_mode_d": 0, 
        "reg_ch_current": 1.0, 
        "reg_ch_mode": 0, 
        "reg_ch_volt": 12.0, 
        "regen_discharge_mode": 0, 
        "restore_current": 0.1, 
        "restore_time": 3, 
        "restore_voltage": 0.5, 
        "run_counter": 0, 
        "safety_cap_c": 120, 
        "safety_cap_d": 120, 
        "safety_temp_c": 45.0, 
        "safety_temp_d": 45.0, 
        "safety_time_c": 0, 
        "safety_time_d": 0, 
        "save_to_sd": true, 
        "store_compensation": 10, 
        "type": 6, 
        "type_str": null, 
        "use_flag": 0
    }
]

johncclayton, 2017-01-10:
FYI the use_flag is the determination here. 0 means fixed, 0x55aa means in use and 0xffff means hidden. 

I'm not distinguishing between signed / unsigned so we should see negative values here for a hidden setting.

scornflake, 2017-01-12:
The reason the name is empty is because it's a preset that has just been made. By default, unless you give it a specific name, the iCharger will "make up" a name based on type, current and charge mode. If you specify a name all that goes away and you get your name. This is now somewhat replicated on the client.

closed 2017-01-12 by scornflake

johncclayton, 2017-01-12:
Aha!  Shouldn't I replicate that on the server instead and hide this impl detail?

scornflake, 2017-01-12:
Not sure you should. It’s almost like a “description” method on an obj-c class. No big deal.

—
Neil Clayton
neil@cloudnine.net.nz

scornflake, 2017-01-13:
So, did some playing with docker today (on a feature branch).

Got it working, and I’ve put some scripts in there to make it simple enough.
I uploaded to hub, see scornflake/electric-pi.

That pulls down, runs (I used restart policy to get it to restart on reboot), and can talk to the charger! Wheeee.
Now I’m turning it off, cos, faster to do a pull myself really locally. Might work on a simple startup script for hypriot next, so I don’t have to log in all the time.

How’d you go your auto git pull?

—
Neil Clayton
neil@cloudnine.net.nz

johncclayton, 2017-01-13:
You mean how'd I do automated builds based on GitHub pushes?

Thanks,
--
John Clayton
Skype: johncclayton