johncclayton/electric #37: Is there a way to tell if the charger is set to metric or imperial (°C or F)?

scornflake, 2017-01-12:
I think we might want to provide the possibility to override this on the front end (maybe?) or just go with whatever is setup in the charger?  We need to know what that is.

For now, I can just have a setting on the config. I figure I'll just put a °C or F after any temp value. No conversion. If the charger is configured metric, and front end is set to imperial, you'd just end up with the wrong label on the end.

johncclayton, 2017-01-19:
yeah, that setting is derived from a 0 or 1 value - and i just transform that to C or F.  You could rely on that for the units I suppose we'd need to test what happens to the numbers when you change the h/ware to F ...

scornflake, 2017-01-19:
is that exposed someplace in the JSON?
as a … var … with … a name?

—
Neil Clayton
neil@cloudnine.net.nz

scornflake, 2017-08-10:
- We already show presets as C & F (we always show both)
- Niiiiice, pipes can participate in dependency injection... I solved this one by adding a "temp" pipe, which does nothing if config is in C, and converts if it's in F.  Yay! All code remains in C, and conversion happens on display only. Woot.

johncclayton, 2017-08-14:
worketh for me.

closed 2017-08-14 by johncclayton