johncclayton/electric #47: Architecture change: poll on the server

scornflake, 2017-01-15:
Thinking about a change whereby we poll all required info on the server, and cache that on the web front end.  That way multiple clients don't introduce many requests over USB.

I think the charger only updates it's internal state about every 250ms max anyway (i.e: when charging, you don't see things change quicker than this).

I did try to get updates faster at one point, but I'm not sure if that failed because a) the USB stuff is broken, b) concurrency limits at the moment or c) it really doesn't provide information at a period < ~250ms.

Plan:
- Have the server auto poll the charger when any of stats/channel/ (other) calls are made. It stops polling "n" seconds after the last poll request. i.e: 10s.
- Cache all those results
- status/channel/(etc) then return data from the cache

This would have a second nice side effect, that all data would update at the same on the ui.  The HTTP side could then be a multithreaded/whatever as heck.

I havn't thought much about how technically we might do this, because I'm sure there are options (schedulers, queues, mice or even superheroes).

johncclayton, 2017-01-19:
Why do this?  What's the root motivation?

scornflake, 2017-01-19:
Merely that the charger itself has a fixed bandwidth, which is even less when charging (we think, maybe, assuming it’s blocking/disallowing calls about 1fps while ‘doing something’).

Yeh, its really, really low priority and only something Id really consider if we have a problem with throughput causing UI issues.

—
Neil Clayton
neil@cloudnine.net.nz

scornflake, 2017-01-30:
I'm gonna say: Naaaaah.

closed 2017-01-30 by scornflake