johncclayton/electric #49: Channel 2 shows high voltage inputs - but there is nothing connected to it

johncclayton, 2017-01-19:
[image]

scornflake, 2017-01-19:
OK. The reason is that we use the per channel amp max to work out if we should show it or not.
Expectations were that the channel would be showing +40v in the case of the 410.

I need a couple of things from you:
a) What is the ‘pack voltage’ without the main charge lead connected, but with balance leads connected?
b) What is the ‘pack voltage’ then, without the balance leads disconnected? 

—
Neil Clayton
neil@cloudnine.net.nz

scornflake, 2017-08-09:
oi. reply to this one please! gimme some data :)
Does it still happen for you?

closed 2017-08-19 by johncclayton