johncclayton/electric #51: UI change detection (when editing presets) could be better

scornflake, 2017-01-30:
Ionic2 is making numbers -> strings, so the change detection sees changes where they don’t REALLY exist (say, you type 0 into a field that already has 0 in it)

See: preset.ts, ionViewCanLeave()

Using lodash _.isEqual, which also sees changes to types.
We want to relax this somehow, or coerce numbers to numbers. Keeping the right type.

scornflake, 2017-08-10:
ok. this is a bit evil.

It turns out that Ionic (not ng2) pretty much casts input to a string when doing the data binding to the model. Thus even though the js code declares a setter as taking a **number** it sends a string.

Because our code is kinda dumb, it takes that value and puts it into the json. 
This DOES Actually work on the server, as values are eventually coerced by schematics (I think).

anyway. It causes the existing preset "have I changed" comparator to fail because types change, and lodash sees that as a change.

I've worked around it by manually coercing every single number (e.g: change thing = value, to thing = +value), which I presume is the same as saying thing = 0 + value.

It's not nice. It's error prone for future code...
Thoughts?

Maybe run schematics or similar on the js model?
How much do we care?

johncclayton, 2017-08-10:
I'd ask on the ionic forum why this is the case - especially given typescript and types are supposed to help instead of being cast away like this.

johncclayton, 2017-08-10:
Another idea - what if you asked the server whether or not the data has changed? Then we can likely re-coerce and use the python code to check for changes?

scornflake, 2017-08-18:
im gonna go with what we have. yeh, it could be better (cleaner on the UI side, less "Oh I forgot to add a +", but... meh.

closed 2017-08-18 by scornflake