johncclayton/electric #62: Build jobs for everything

johncclayton, 2017-06-05:
This is probably worth doing:
http://docs.ionic.io/services/deploy/

Ideally - iOS and Android build jobs + automated Docker + SD-CARD

scornflake, 2017-06-05:
duplicate of #53?

johncclayton, 2018-04-15:
Dupe, I don't think so.  I like the idea.

johncclayton, 2018-04-15:
This morning I'll look at making sure the configuration for the sd-image building process is driven entirely from the branch such that we can adapt it - we need that so that the existing production builds can continue working OK while at the same time I can mess with a branch that builds things differently for the rpi3b+

johncclayton, 2018-04-21:
Changed the intent - the build jobs for branches are now done.  I'd like this ticket to now simply target having a build job for Android