johncclayton/electric #75: Integrate network lifecycle with the app better

scornflake, 2017-08-09:
Example:

the charger service will modify the configuration (celsius flag) after the configuration is loaded (see     private chargerDidAppear(statusDict))

But is this the right place? I don't feel like lifecycle is well defined from a network connection / what the app then does point of view.

- Say you're sitting in presets, and the network dies. What should happen?
- Say you're listing presets, and you can't contact the server... what should the UI show? Should it retry? Should that be automatic?  
- What does complete failure do?
  - Bring us back to a "Connecting ... " page?
  - Just retry automatically?

johncclayton, 2017-08-18:
I would want to close all dialogs and go back to a connecting... page / lost connection page.  There is no sane way to resume anyway; so don't try - make it clear that a connection is required to the charger (worker).  

We still may need to ensure that temporarily failed requests don't cause this logic to kick in; i.e. how long does a connection problem need to exist before the state of "disconnected" kicks in.

scornflake, 2017-08-19:
- if on home screen (volts, it), network connection will cause "looking for charger"
- if on presets or config, it'll stay there with a toast warning

closed 2017-08-19 by scornflake