johncclayton/electric #79: Move measureIR fully into the server (inc. timeout / stop)

scornflake, 2017-08-15:
The measureIR server function should auto timeout itself after seeing some amps on the channel.

This is what the UI used to do.  It'd kick off a low amp discharge, and wait to see some amps. It'd then stop the operation.  

However; I feel it is probably safer to do this in the server.
But: this will then up being a blocking call.

Thoughts?

What I don't want to do is kick off a measure, and have the possibility of it running for ages.
I guess worst case, it'll discharge the pack and the charger will stop.

johncclayton, 2017-08-15:
How about we provide the POST resource call to kick it off, and for now we hard-code how many seconds it'll run for.   I would make it an async call though.

The status and/or run-time information/state associated with IR measurements can be made available within the existing channel information or as a separate REST call.  I prefer the former along with a value that indicates whether or not the IR measurement is actually running.

I do like the idea of putting this in the server because it means multiple requests to measure IR can be ignored/cascaded; and any action on the channel where IR is being measured can be ignored or it can replace the IR measurement safely, because the server can better check charger state. 

The existing code in the server can be easily adapted to cancel an IR measurement on a channel.

scornflake, 2017-08-15:
I’ll show you how I’ve done it on the client, and we can discuss further…

johncclayton, 2017-08-16:
[image] [Discuss how to do measureIR](https://trello.com/c/BBfJb06u/2-discuss-how-to-do-measureir)

johncclayton, 2017-08-18:
Closing in favour of a more accurate description of what we really want to achieve - namely #88

closed 2017-08-18 by johncclayton