johncclayton/electric #86: Exception when dismissing connection warning

scornflake, 2017-08-18:
1. Have an IP that isn't valid (doesn't connect to a Pi)
1. Wait for connection error to show
1. Dismiss

Should end up with:

johncclayton, 2017-08-18:
Error
Close
Runtime Error
Uncaught (in promise): inserted view was already destroyed
Stack
Error: Uncaught (in promise): inserted view was already destroyed
    at c (http://localhost:8100/build/polyfills.js:3:13535)
    at Object.reject (http://localhost:8100/build/polyfills.js:3:12891)
    at OverlayPortal.NavControllerBase._fireError (http://localhost:8100/build/vendor.js:61866:16)
    at OverlayPortal.NavControllerBase._failed (http://localhost:8100/build/vendor.js:61854:14)
    at http://localhost:8100/build/vendor.js:61909:59
    at t.invoke (http://localhost:8100/build/polyfills.js:3:9283)
    at Object.onInvoke (http://localhost:8100/build/vendor.js:4509:37)
    at t.invoke (http://localhost:8100/build/polyfills.js:3:9223)
    at r.run (http://localhost:8100/build/polyfills.js:3:4452)
    at http://localhost:8100/build/polyfills.js:3:14076
Ionic Framework: 3.6.0
Ionic App Scripts: 2.1.4
Angular Core: 4.1.3
Angular Compiler CLI: 4.1.3
Node: 7.10.0
OS Platform: Windows 10
Navigator Platform: Win32
User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.101 Sa

scornflake, 2017-08-19:
- press dismiss
- receive a 'onDidDismiss' where we cleanup the obj.
- I think setting the object to nil prematurely is causing fault. lets try leaving it around for GC.

scornflake, 2017-08-19:
Yeeeeeehs. That's it. Don't mess with Mr Object.

closed 2017-08-19 by scornflake