johncclayton/electric #90: Worker taking up 100% CPU and growing RAM footprint

johncclayton, 2017-08-18:
This occurred when running the worker and detaching the charger.  Needs investigation.

scornflake, 2017-08-19:
I have seen the server suddenly queue a LOT of requests when I've reconnected it. I don't yet know where they are being queued (flash or zmq, or which side of that). Looks like some kind of backing up of request is occurring someplace tho. Not sure it's related to this?

scornflake, 2017-08-19:
I've added some exception handling to:
    def _send_message_get_response(self, method_name, arg_dict=None):

the send of the request was failing.  I've done the same thing, and reset the connection.
I don't know if this is the *right* thing to do tho.

closed 2017-08-19 by johncclayton