johncclayton/electric #94: Balance leads / pack state on UI isn't quite right

johncclayton, 2017-08-19:
When plugging in balance leads with pack already plugged in, UI goes to 'no pack' before going 'Idle', why?  This should actually be a single state thing

johncclayton, 2017-08-19:
When the balance leads are plugged in, sometimes (and lord knows why, that's Junsi), the cell values are not entirely populated. 

Potential solution: analyze the number of connected cells over a few samples and only then make a decision as to whether a pack is connected.

johncclayton, 2017-08-19:
pcnt_diff = 0
        if self.cell_count_with_voltage_values > 0:
            pcnt_diff = abs(100.0 - ((self.curr_out_volts / self.cell_total_voltage) * 100.0))
            # if the difference is too great - then we think that curr_out_volts is total lies
            if pcnt_diff < 3.0:
                plugged_in = True

scornflake, 2017-08-20:
So this is done in the client now. But it gets worse.

In the case of the 2s pack being charged with no balance plugs (a special preset setup, saying 'dont balance, it's 2s'), when plugging in... we STILL show "no pack".

I'm going to take out the "balance leads" text, and simply go direct to idle.
The "basic" voltage check has been simplified to ignore the number of connected cells. Now, it just uses device_id, and if volts is < expected, assumed a pack is connected.

The more accurate check is still there, but it only happens if balance leads are connected anyway...

scornflake, 2017-08-20:
Right. I think this is done now.

closed 2017-08-20 by scornflake