johncclayton/electric #95: Can't start or stop a charge without balance lead in the app.

JoakimSelander, 2017-08-19:
Some batteries like the Jeti receiver batteries don't have a balance lead. It should be possible to charge these anyway. 
Best would be to have a "Are you sure you want to charge without balance?" popup.

scornflake, 2017-08-20:
how is such a pack normally charged? Does the charger even let you do it?

scornflake, 2017-08-20:
Fixed. You have to setup a preset that'll let you do this in the first place (disable balance checks and set the cell count correctly). Now it'll work.

closed 2017-08-20 by scornflake

scornflake, 2017-08-20:
> Well it doesn't work for me. Edited the preset and saved it. When I press on the cells it just says "Pack not plugged in. So I can never get to selecting the preset.

If you’re not pulling from github, the existing pi builds prob won’t work.
GitHub will reflect the most recent dev commits, but unsure if you’ve done a direct git clone or running a built instance. If later, there’ll likely be no chance methinks… we’re still sorting build/deploy.

Based on the message you got, I’d say you’re still running the older codebase.
 
How technical are you? I’m happy to run through some setup with you / teamviewer/skype if you want.
At this stage, we know for sure that it’s not user friendly!

—
Neil Clayton
neil@cloudnine.net.nz

JoakimSelander, 2017-08-20:
Oh sorry, I've not updated since yesterday.
Also I'm using Ionic view with an ID that John gave me yesterday.
Perhaps you could publish the latest ID somewhere?

johncclayton, 2017-08-20:
Good feedback for us - we need to have a single way of updating both server and mobile apps otherwise it's gonna be ... "interesting" from a support point of view.

scornflake, 2017-08-20:
Sure, I can do that, I’ll push one now.
Done.

—
Neil Clayton
neil@cloudnine.net.nz