johncclayton/electric #111: Update for 4010 Duo

hafromue, 2018-01-28:
Hello Neil,
when can you expect an update for the 4010duo?
I can not call the "Presets" on my 4010duo and start nothing.

Regards
Alex

scornflake, 2018-01-28:
Can you give more detail? John has a 4010 and he did development of the app with me. Wondering if something else is up?

—
Neil Clayton
neil@cloudnine.net.nz

hafromue, 2018-01-29:
Hello Neil,

I have a Samsung tablet and a HTC smartphone. Both devices with Android. I can see all the values displayed on the start page.
I can also stop a process. But I can not start any processes.

mit freundlichen Grüßen / with best regards

Alexander Fromm
Fasanenring 19
86666 Burgheim
Tel.: +49 (0) 8432 - 263981
Fax: +49 (0) 8432 - 263985
Mobil: +49 (0) 1522 - 9207700

neil@cloudnine.net.nz<mailto:neil@cloudnine.net.nz>

scornflake, 2018-01-30:
What happens if you tap “a lot” on the cells? Does it eventually bring up a menu?

—
Neil Clayton
neil@cloudnine.net.nz

hafromue, 2018-02-01:
Hi,

if a process is running and I tap twice on a cell field then a menu opens and I can stop this process.
If there is no process and there is a battery connected and I tap twice on a cell array, a menu opens with all sorts of operations but I can not start any of the processes.
The predefined batteries are also not displayed.

scornflake, 2018-02-01:
Hmm. Sounds like something John had happen with his 4010. I’ve CCed him in. John, does any of this ring a bell?  i.e: unresponsive UI, no presets, etc, with the 4010?  Thinking about that power isolator you had to use…

—
Neil Clayton
neil@cloudnine.net.nz

johncclayton, 2018-02-05:
Sorry for the delay in my reply - I've been away on holiday and also sick (wahaay - that garners COMBO points).  

Specifically, Neil mentions the 100mAh USB isolator: https://docs.google.com/document/d/12vy4kCue40k26qsqJIa6b5kwuOIhKOWrTJteruaGcJk/edit#bookmark=id.f1c7xn347c2x

I found that I absolutely needed it.  It sits between the Pi and the iCharger.  Without this unit the Pi wouldn't connect correctly to the iCharger; it was as if the USB port was dead.  I'm not sure this is what you are seeing however. 

My Pi is running from what I, as an unqualified electric guy, would assume is a single power source.  240v goes into the charging case.  Step down regulators then feed 5v into the Pi off this power source, and the same source feeds the iCharger directly.  I figure that DOES NOT create a ground loop, but what do I know (not enough to verify this assumption - I can assure you). 

--
John

scornflake, 2018-03-02:
@hafromue any news on this from your end? did you try the isolator?

hafromue, 2018-03-02:
Hello Neil,

I ordered an insulator but it has not arrived yet.
I will tell if it works when the insulator has arrived.

Regards

Alex

hafromue, 2018-03-02:
Hello,

I ordered an insulator but it has not arrived yet.
I will tell if it works when the insulator has arrived.

thomasekdahlN, 2018-03-03:
@johncclayton - About powering the 4010. I have tried powering the Pi from the main power cables of the 4010 (running 12-24) volts, putting the step-down between (to try to have it work without 220v) but only on same power source as charger. It looked like a good idea, but I have not been able to feed the Pi with power directly using +/- and fint it to difficult to solder the step down to a USB cable. Will do some more research on feeding the Pi directly from the regulator (soldered in).

scornflake, 2018-03-03:
Another option you might consider: I ended up powering my pi3 using a BEC that I had laying around, that’d output 5.1v or so.

I cut up a USB lead and rewired that to the output of the BEC, after checking that the pi3 was going to be OK with 5.15v (spec says 5v… I wasn’t sure if it’d survive, but it seems to be OK, it’s been running for months now).

—
Neil Clayton
neil@cloudnine.net.nz

thomasekdahlN, 2018-03-03:
Thanks. The step-down I mentioned is a Bec;-) I'll try that, sounds easier.

scornflake, 2018-03-04:
> Do you also use the USB isolator component in between in this setup?

I personally didn’t need to (I’m running a 308 duo tho).  

I did  originally run the pi3 from the duo itself, but I was seeing ‘the electrical symbol’ top right on screen (via HDMI) when testing the pi3, meaning: not enough juice. I was also at that time experiencing weirdness. After switching to the BEC, no problems.

—
Neil Clayton
neil@cloudnine.net.nz

johncclayton, 2018-04-15:
Hi - any progress on this?  Did it work?  The symptoms are very similar to what I experienced without an isolator.  Putting this little thing between the USB port on the Pi and the communication/USB port of the iCharger resolved it: https://www.adafruit.com/product/2107

thomasekdahlN, 2018-04-15:
Yes it worked without isolator, soldering the USB cable to the BEC - soldered to the main input cables to the DUO. But I'm having trouble with the latest image so the project is not running at the moment.

johncclayton, 2018-04-15:
I'm messing with the builds right now.  I realize we need a way of keeping
builds around that are stable / known to work, this is something I'm trying
to address today.

Thanks,
--
John Clayton
Skype: johncclayton

johncclayton, 2018-05-22:
> Now the Isolator arrived.
> I tried it but no changes!!

Can you show me how you wired it up please?

John Clayton
CTO
E: johnc@filewave.com
A: St. Gallerstr 1, 9500 Wil, St Gallen
<https://maps.google.com/?q=St.%20Gallestr%201%2C%209500%20Wil%2C%20St%20Gallen>
W: https://www.filewave.com

scornflake, 2018-11-08:
any activity, going to close shortly if not