johncclayton/electric #113: Feature request: Support for rc-thoughts RFID to start charging

thomasekdahlN, 2018-02-18:
Love your project, using it on my 4010.

What would really top it off, was to configure all battery charging parameters by reading the rc-thougts RFID, that contains: 

Tag  ID, Battery Capacity, Cycles-count, C-value and Cellcount
https://www.rc-thoughts.com/jeti-rfid-sensor/

Idea:
- read RFID tag to set alle chargin parameters for the battery correct (for 1C charge)
- bump the Cycle-count on each charge - keeping track of the number of times each battery has been charged
- Maybe also add internal resistance data to bump tag if possible.

TheBum, 2018-02-21:
It's feasible, but it would be best to eliminate the Arduino and let the Pi do everything. I found this page for interfacing an RC522 RFID reader with a Raspberry Pi that would work alongside the case fan control circuit. I may give it a shot once I finish my other ongoing projects.

https://pimylifeup.com/raspberry-pi-rfid-rc522/

TheBum, 2018-02-22:
I have an RC522 RFID on order from Amazon. It should arrive tomorrow.

thomasekdahlN, 2018-02-22:
Great. Looking forward to the results. I agree (and Tero) on coupling the RFID reader directly to the PI (No Arduino). This will make your solution the very best. I love it.

thomasekdahlN, 2018-02-22:
If you need help reading/understanding the tag format, I'm sure Tero will help.

TheBum, 2018-02-22:
That's something I'll have to research once I figure out how to get the reader and the Pi to talk with each other.

RC-Thoughts, 2018-02-27:
I think I have a extra RasPi 3 somewhere, let's have some fun! 👍

TheBum, 2018-02-28:
I played around with some code for reading specific blocks using lower level routines; the higher-level routines in the sample code always deal with text in blocks 8-10. When sending the IDLE request at the beginning of the sequence, it returns a TagType value that may or may not be what I need to distinguish between the Revo and generic tags (there's absolutely no documentation with this code). The value it returned for the keychain FOB included with the reader/writer was 0x10, which according to the Arduino library code translates to a MIFARE Plus. I'll probably have to wait for my MIFARE 1K tags to arrive to verify.

I'm designing tag read polling to run in its own thread, under the assumption that the system should be ready to read at all times when not writing a tag. Once the data has been parsed, the system will set a flag that the mobile app can key on during its polls of the Pi to pull the battery parameters across. The mobile app will send a request to reset the flag after it has obtained the parameters (or it can be reset automatically after responding to the parameter request) and then present an option of what to do with that pack (charge, discharge, storage charge, balance), maybe in a simple grid of buttons.

When writing a tag, the read polling will be temporarily halted.

Thoughts? Am I way off base on the interaction between the mobile app and the Pi or any other part of the implementation?

Also, what constitutes a cycle for incrementing the count? Only charges? Storage charges if the battery is discharged below storage voltage?

scornflake, 2018-02-28:
Yep, a couple off the top of my head:

a) I wonder if we should involve Tero in this conversation? He’s got considerable experience with both Revo and “other” tags and might be able to give us a leg-up in the implementation. Also, I wonder if we could use the same tag layout?

b) When I designed the server/client, one thing that became apparent pretty early on is to have the client being ‘read only’ as much as possible.
It makes coding it far easier, and you can always see the server as a “the truth” (you don’t have to worry about syncing stage, so much).

Let me think here… ponder ponder:

- So, the tag will have info. I like the idea of a thread on the server, polling for that - that makes sense.
- A tag is read, the info goes into the server.  
  - The server enters a “Begin charge with RFID” state.
  - Each subsequent tag causes an ‘accumulation’ of information, e.g: total capacity,
  - We could also do verification at this point (don’t allow tags of differing cells)
  - The resulting profile would be a mix of # packs and min mAh (so, charge at min mAh X # packs, or choose a preset close to that)
- The client sees the ‘Begin charge with RFID’ state, and brings up a UI saying “N packs, max M mAh. Confirm?” (or other UI to allow selection of a preset)
- The server kicks off the charge.

At that point, the client will be reading server side state like it does now.
i..e: I wrote the client/server so the client would mostly (baring #88) be able to simply read data from the server to show the correct UI. That’s also why you can start a charge (or other op) on the hardware itself, and the client responds mostly OK.

So, I think you’ll real close, just I would look to minimise the back/forth between the server and the client as much as possible.

c) Counting

I’m assuming you want to write the charge count back to the tag.

I’d say go simple to begin with. If a charge is performed and takes longer than some time, then we increment.  However; that implies the pack is still near the reader so that we can write the tag, which in my case it wouldn’t be.  I might have 8 packs in parallel, strewn across a large desk area.

So, does that then mean we’d need to increment upon touching the tag?  That’d work most times for me. It’s rare that I’d be touching a tag to setup a charge profile, and then NOT using that profile.  

Another option would be to never write back to the tag, introduce persistence into the server and count the tags there. That’d make incrementing the counter more reliable (we can increment only when the charge is complete, or after a time, or both). But it means you loose the tracking on the tag itself.  I don’t like this idea for the amount of work it introduces tho.

—
Neil Clayton
neil@cloudnine.net.nz

TheBum, 2018-02-28:
I've been PM'ing Tero on HeliFreak. I looked at the Bump Controller user's manual and there are several parameters in the bump tag and missing from Tero's implementation that would be needed for charging. He's following this project (see the RC-Thoughts post above), so hopefully he'll get engaged. I'm attempting to remain compatible with Revo and Robbe.

The required but missing tag values are: Chemistry, Charge Rate (Revo has Accurate, Normal, and Fast rates), and Discharge Rate. I'm hoping he knows where these values reside, or at least can find out. Others values are defined (e.g. termination voltages for charge and discharge, measured cell IR, measured capacity) that we may or may not want to support. We apparently cannot write to Revo tags, so the analytical type data would be limited to the generic tags.

TheBum, 2018-02-28:
> Another option would be to never write back to the tag, introduce persistence into the server and count the tags there. That’d make incrementing the counter more reliable (we can increment only when [...]

We could use a hybrid approach: cache the count on the server and then, on the bump to start the next charge, write the previous count. That would ensure the pack was a maximum of one cycle behind.

RC-Thoughts, 2018-02-28:
Regarding when to update cycles or if they are stored on the server. My sensor has an option to add it in flight (after ~30 sec powered) or it can be disabled if it is updated by charger, so it that sense the sensor is ready. However, if it's stored on server it'll be miss-synced at some point, so my point of view is that it should stay on the tag. Hybrid solution is tempting, that would a) ensure cycle counting and b) reflect "true usage" since the cycle would be added after flight before next flight and on top of that c) be stored on the tag. If it's not on the tag it would only worsen the RFID-information for Jeti-users.

Adding more fields as Chemistry, Charge Rate and Discharge Rate is in no sense any issue even if my sensor does not use them, there's plenty of space on the tag for that. Also if user chooses to use Electric/iCharger/RFID-Sensor combo we do not have to think about using same datablocks as others. And we still are not losing compatibility with Robbe and Revo. (What Robbe-compatibility are you thinking of? Are going to control Robbe-chargers too?)

What comes to Revo datablocks, I'll send you an email.

TheBum, 2018-03-01:
> (What Robbe-compatibility are you thinking of? Are going to control Robbe-chargers too?)

No, just don't want to break the tags in such a way that the Robbe chargers can't use them.

Regarding the hybrid solution, all you'd have to do to get the counts fully synced is to bump the battery on the sensor and then say "no" to charging before you leave the field. You could even tap them all and say "no" only once if we add logic to update the counts regardless of whether the charger rejects a bump due to a cell count mismatch.

RC-Thoughts, 2018-03-01:
Robbe BID code is [here](https://www.onki.de/uebermich/bauberichte/288-bid2nfc-drahtlos-bid-chip-fuer-robbe-lader).

The memory-layout is copied straight from the commercial BID-chip. It uses a big amount of blocks for little data. Not really wise itself. But even with more values (like chemistry etc) there's room for quite a lot more.

TheBum, 2018-03-02:
Making progress on design (I'm really bored at work). I decided that I needed a separate thread for writing too, since it could be a while between initiating the write and bumping the tag. It also allows the write to be easily aborted.

I'll take a look at the Robbe BID code when I get a chance. Any luck locating the rest of the Revo tag information, Tero?

scornflake, 2018-03-02:
Tero has provided me some info / code. I’ve asked him if he thinks it is OK to post here (some of it was from Revo).  I will likely ask Revo their opinion as well. 

—
Neil Clayton
neil@cloudnine.net.nz

scornflake, 2018-03-02:
@TheBum might I suggest a queue rather than multiple threads. You'll likely have  more control, and it'll be easier to debug and synchronize access to the hardware.

TheBum, 2018-03-02:
It's essentially a state machine: either it's reading (99.9% of the time) or it's writing. I've designed it so that only one thread can run at a time. When a write is requested, the read thread is terminated;  the write request is essentially an interrupt of the normal reading behavior. The write thread only runs for as long as it takes to write the tag after requesting it, which includes the time it takes for the user to bump the tag. The read thread won't immediately restart so that the pack can be removed from the sensor after writing without being picked up as a read bump, probably after a user acknowledgment.

TheBum, 2018-03-02:
@RC-Thoughts I'm not seeing any of the mappings to blocks 60 and 61 in the BID code that I saw in your code. I'm probably just missing some context, especially since the site is in German and Google Translate doesn't do a perfect job, especially when there are typos.

RC-Thoughts, 2018-03-02:
BID uses one or two bytes here, another there, bytes used are "all over the place" but BID does not use blocks 60/61. I do for the reason you guessed, 60/61 is not used in BID, therefore my tags can be "dual-used" as BID-tags also. This was a request from some German users. Since the Robbe BID is also DIY I have no idea of amount of users', would guess pretty low. RCT-sensors are used by few hundreds users to my knowledge based on surveys, sold sensors and forum-traffic on subject.

And yes, for the chemistry etc for example 62 is usable. (May I suggest numerical form in tag instead of ASCII like "LiPo", this way I could use it too later on?, For example 1=LiPo, 2=LiFe etc. I cannot use text due Telemetry restrictions)

johncclayton, 2018-03-02:
Try DeepL instead of Google - it's WAAAAY better at translation.  Or ask me
- I can give the German stuff a shot.

Thanks,
--
John Clayton
Skype: johncclayton

TheBum, 2018-03-02:
No real need now. Now that I know Tero used blocks not used by Robbe, then the Robbe format is immaterial.

It does beg the questions:

1. Why did you only use 8 bytes of the 16 in block 60?

2. Why didn’t you keep the cycle count instead of the battery C rating in block 61 with the “disable counts” field?

RC-Thoughts, 2018-03-02:
1. Because I didn't know better, RFID was the first sensor I ever made as a total programming-n00b (still am) and RFID-part is for the most part a copy from another project by Alastair as credited in notes. And it have worked for 1.5 years :)

2. I don't understand, what do you mean? Actually I found a missing write-byte on block 61 when cycle-count is increased when I looked at the code trying to understand the question, is it this you mean?

TheBum, 2018-03-02:
1. Understand. I forget that some people are new to big software projects. I've been developing software for 26 years.

2. I just meant that I thought the stuff that's more or less static would be on one block and the more dynamic stuff (cycle counts and whether or not to increment cycles) would be on the other block. I noticed that cycle counts are in block 60 with the static data and C rating (which is static) is in block 61 with the dynamic data.

TheBum, 2018-03-02:
The way I see it, we need to add the following fields to what Tero has defined on the RCT tag:

1. Maximum charge rate as a C multiple. This is a not-to-exceed value that the bump logic will validate against. This will likely be stored as an unsigned 16-bit integer.

2. Desired charge rate in mA or a multiple thereof. The main question for this field is whether to make it 16- or 32-bit. If using whole mA, a 16-bit unsigned integer would allow a maximum charge rate of 65.535A, While this might be OK for modern times, it might not stand the test of time. So, the secondary question is: should it be stored in 16 bits with a bit resolution of, say, 10mA, or in 32 bits as a whole number of mA?

3. Desired discharge rate in mA or a multiple thereof. For consistency, keep the same integer range/resolution as the desired charge rate.

The maximum discharge rate would be the C rating of the pack, which already has a defined field on the RCT tag. 

It would be nice to know how the Revo tag stores these values. We may want to follow their lead.

scornflake, 2018-03-03:
2) I would have though resolution of 100ma would be fine. When do you really need less than that? Even on a 800mah single cell, that’d give you all the options you’d need, wouldn’t it?

—
Neil Clayton
neil@cloudnine.net.nz

TheBum, 2018-03-03:
I think 100mA resolution would be reasonable too. Does the iCharger even allow charge or discharge currents at a resolution less than 100mA? 

It did get me to thinking about how the capacity is stored on the RCT tags. 20000mAh packs are not uncommon in the drone world and I think the RCT tag format uses whole mAh in a 16-bit number. I could foresee 66000 or larger packs in the near future. The best way around that IMO is to add a version field, maybe in one of the block-60 bytes. If zeros are written to the unused bytes by the Jeti RFID module right now, we could start at version 0. This might be the time to define a version 1 that allows larger capacities than 65535mAh.

scornflake, 2018-03-03:
Or just use 32 bit and be done with it!  Are we short on bytes?

—
Neil Clayton
neil@cloudnine.net.nz

RC-Thoughts, 2018-03-03:
I think this is wrong: "The maximum discharge rate would be the C rating of the pack, which already has a defined field on the RCT tag"

65C or 70C batteries are not uncommon, as charge-rate it's way too high. To my knowledge highest charge-rate from manufacturer is 10C for a 60C battery. (These apply to LiPo)

johncclayton, 2018-03-03:
Time for me to chip in. No pun intended.

I love the idea of the feature but I would like to point out that I've been
made aware (before we started this project) that FMA have in the past taken
legal steps to protect their inventions.

Speficially in the area of using an RFID tag.

So while I totally support the concept of RFID being used I would not want
this in the main branch of our repo.  I would not want the entire project
to face some kind of legal issue for one feature.

I suggest that this feature be segregated via either a plug-in or repo fork
such that if the case arises that legal steps are taken then the main
functionality doesn't have to be pulled out and can still benefit everyone.

I'm totally open to discussing how we might do that.

Thanks
--
John Clayton
CTO

RC-Thoughts, 2018-03-03:
I just wrote to Revo and asked it straight out.

TheBum, 2018-03-03:
> I think this is wrong: "The maximum discharge rate would be the C rating of the pack, which already has a defined field on the RCT tag"
>
> 65C or 70C batteries are not uncommon, as charge-rate it's [...]

Note that I’m talking about maximum *discharge* rate, i.e. the rate at which the pack can be run down, not charged. A 60C 5000mAh pack may only be able to be charged at 25-50A (5C-10C), but it can be discharged at 300A. It may not have a long life if run that hard, but it’s spec’ed for that. 

As for legal ramifications, I’m OK with a fork of the server, but there has to be a user interface for the functionality to be useful. You can side-load an app in Android, but not in iOS; the app has to be released to the public except for test flighting, which has a limited lifetime.

I don’t know how NZ IP law is, but in the US, reverse engineering is legal except for cracking encryption. As long as you’re not stealing their source code or screen layouts, you can do your own implementation that has similar features. You’re not supposed to be able to patent an idea, only the implementation. If we leave support for Revo tags out completely, there’s not much that Revolectrix/FMS can do.

TheBum, 2018-03-07:
I have the code working to write and read RCT tags as well as accumulate charge and discharge rates for multiple packs. I have uploaded the files to my fork (rfidtagio.py, test_tag_reader.py, test_tag_writer.py). It still needs more testing, but it's at least 90% stable. I've programmed two tags and tested them with multi-tag reading.

TheBum, 2018-03-10:
Is there any way to send instant notifications from the server to the mobile app? If not, I may add a 3-color LED to a couple of GPIO outputs provide an indication that a tag scanned or was rejected (e.g. if it was already scanned in the current parallel set, or the tag hasn't been written to yet).

TheBum, 2018-03-12:
I think the server changes are done. The conventions for the LED indicator are:

- Solid green = reader active, ready to accept tag touches.
- Solid amber = writer active, waiting for the tag to be touched for writing.
- Green flashing (4 times) = tag registered in the tag list. All registered tags must have the same chemistry and cell count.
- Amber flashing (4 times) = tag successfully written.
- Solid red when reading = Tag rejected: tag data is not valid, or chemistry or cell count don't match.
- Solid red when writing = Tag has already been written and the "force" option wasn't specified, or tag writing may have failed (tag couldn't be read after the write attempt).
- Off = RFID tasks are idle (no thread running).

There is one case where I'm not sure what to indicate:

- When reading, what should be indicated if a tag is touched that had already been registered during the current read session? Options are: stay solid green (no indication), green flashing without adding again to the list, or red to indicate that it cannot be added to the list because it's already there. Maybe a different indication?

TheBum, 2018-03-12:
I did a little video demonstrating the operation using Safari and curl. I uploaded it to YouTube as an unlisted video so that it’s not accessible to the general public unless they have the link. I can either post the link here, or send it to interested people via email or IM/PM.

tsteneker, 2018-03-12:
I am interested Great stuff you are doing.ThanksTerry

RC-Thoughts, 2018-03-12:
@TheBum please send the link, very curious :)

scornflake, 2018-03-12:
Im curious too!
I’d say, lets post the link here. We’re going for an open implementation, so I don’t see a problem sharing tests / attempts / working variants (unless… I’m missing something super secret special?)

—
Neil Clayton
neil@cloudnine.net.nz

johncclayton, 2018-03-12:
Hi Guys

While I'm super being with my day job (and life in general), I'd love the
link as well in order to see progress.  The ideas here are awesome and I'm
glad that the open source model is benefiting more people than we'd
otherwise reach.  I'm not currently searching the good 'ol net for activity
here - are you gents also active on helifreak or something other channel?

Thanks,
--
John Clayton
Skype: johncclayton

TheBum, 2018-03-12:
I just wasn’t sure how much attention we wanted to draw from Revo with an actual implementation, in case they have a problem with this. Here’s the link:

https://youtu.be/6x2ptB1sdSo

Some discussion has taken place in Neil’s thread on HeliFreak, but for the same reason I’ve been reluctant to post the video link, I’ve avoided posting anything about it there. We can certainly move the discussion there.

johncclayton, 2018-03-12:
WOW on the RFID function, very nicely done!

I think the easiest way is likely just to ask them.

If... no... when this get published they'll find out anyway.

Using a JETI RFID sensor is potentially less risky from a legal point of
view as it's not related to charging gear - yet hooking RFID directly to
the charger and driving interactions with is maybe/likely/going to raise
FMA's eyebrows - if I had a patent on that, it would raise mine.  If they
do pull something like a cease and desist then what does one do?  Does
anyone here have contacts with a patent lawyer that could guide a first
contact?  For example - it could well be that a cease and desist is just
hot air designed to scare.  For example, if a C&D is delivered - must it be
presented alongside a publically visible patent in certain countries?  I
have no idea, I'm not a patent lawyer but I could easily believe scare
tactics would be a sensible first port of call for any business as it's
cheap to execute.

So I'd like to know from you guys - do you have any contacts?  I'd be
willing to fund a small amount of legal expense to investigate an initial
C&D attempt by FMA should we get to that point - simply because I don't
want to see a community project get killed by scare tactics (IF that's what
they are).  I'd be interested to talk to you guys about your feelings on
that as well - in the first instance via Skype/Discord/Telegram/whatever.

If you're willing - please contact me on Skype and we'll find a time to
chat.

Thanks,
--
John Clayton
Skype: johncclayton

TheBum, 2018-03-12:
As I mentioned earlier, you’d think if their Bump technology were patented, they’d have “patented” or “patent pending” verbiage on their website. I’ve looked extensively and the only patent I’ve found mentioned is for the Copilot II, presumably for the IR technology they use.

I’m going to make a small tweak to the LED indications: if a tag is rejected as part of an ongoing operation (e.g. when reading tags), the LED will flash red rather than go solid red. Solid red will be reserved for fault conditions that terminate the current operation (e.g. a tag write fails).

TheBum, 2018-03-12:
One last bit of housekeeping: the MFRC522.py module. It's only available in a git repo, i.e. it has no apt-get or pip installer, so how do we go about installing it in the worker docker image? Clone the repo to a temporary location and then do a simple copy of the required file into the image as RUN actions in Dockerfile-worker?

TheBum, 2018-03-13:
Just a few notes of things that the mobile app needs to be aware of:

The requests for the RFID _reader_ are all JSON blobs PUT to the rfid/read subpath of the following form: { "command":<command string>" }.

- Start reading command string: "start reading".
- Stop reading command string: "stop reading".
- Kill reading command string: "kill reading".

The difference between "stop" and "kill" are that "stop" keeps the thread object alive that contains the list of tags read so that a GET can be issued to retrieve the list without the possibility of another tag being scanned. Kill will destroy the thread object; doing a GET after a kill will always give you an empty tag list and a "dead" status.

TheBum, 2018-03-13:
Since no RFID tag reading or writing will be initiated without the mobile app sending a command first, I think we could go ahead and merge these changes into the master. Let me know and I'll create a pull request.

TheBum, 2018-03-15:
Battery cycle tracking is done. You all (or all y'all, in Southern vernacular) may beta test away.

thomasekdahlN, 2018-04-01:
Great work. I would love to beta test.
1. Do you have a component list and wiring diagram for reading the RFID tags?
2. Which image version is the beta functionality available on?
3.  I'm willing to fund a small amount of legal expense to investigate an initial C&D attempt by FMA - this project is to great to be killed by lawyers;-)

scornflake, 2018-04-01:
> Great work. I would love to beta test.
>
> Do you have a component list and wiring diagram for reading the RFID tags?
> Which image version is the beta functionality available on?
> I'm willing to fund a small amount of legal expense to investigate an initial C&D attempt by FMA - this project is to great to be killed by lawyers;-)

Hi!

Currently I think ‘thebum’ has a working branch with RFID. I’m intending to try to use my existing RFID boards (no idea if that’ll work yet). They are the same as the smaller green ones found on rc-thoughts.com (for the Jeti RFID reader/writer).

I’ve yet to do *anything* on the UI.
There is server code implemented, and the server side can detect things like a batt being read, can sum up further batts being read (i.e: 5 6s have been seen, suggested C of 3, with max mah 3600).  The client doesn’t do anything with this yet. That’s the bit I need to get onto.

Other than that, no idea on the legal side. I know either thebum or (someone?) did email revo directly. I have’t heard of any response yet.
Current intention is to write the UI so it’ll work on both branches. So on the ‘master’ it won’t show RFID. On ’thebums’ branch, it will. That way we have options, if required.  Also, merging it back to master will be a piece. of. cake.

I like cake.

Off to play with my TDF! 

—
Neil Clayton
neil@cloudnine.net.nz

johncclayton, 2018-04-01:
Reading RFID tags: I'll let somebody more knowledgeable (ThBum?) on that
one go for that question.

In terms of legal expense I also don't want to have the project killed. I'm
also able to help out a little on the finance side to push a C&D defense.
I've looked for their patent and haven't been able to find a version of it
online, which I believe doesn't mean it doesn't exist but there I'm not
sure - patent lawyer question.

Does anyone on the group have access or contacts with an experienced patent
lawyer. I can find some but I'm in Switzerland and these guys cost at least
300 per hour.

TheBum, 2018-04-01:
It should work with any RC522 RFID reader/writer. It has to use SPI for communication. I used this one to develop the code:

https://www.amazon.com/dp/B01KFM0XNG

There are multiple resources on the web for connecting it up.

TheBum, 2018-04-10:
Any progress? I know sometimes work gets the best of your free time, so I'll understand.

scornflake, 2018-04-11:
Not so far - I did enter my mind yesterday, but nothing found its way to my fingers

—
Neil Clayton
neil@cloudnine.net.nz

TheBum, 2018-04-13:
Am I interpreting the code correctly that presets can be created without saving them to flash? If so, I think I'm going to experiment with a server-side implementation of RFID charging/discharging using a temporary preset. An "RFID" preset will be built up as tags are scanned. To start the charge, simply select the "RFID" preset. I have to modify the code anyway to support two channels worth of RFID information.

scornflake, 2018-04-15:
From memory I don’t think so. I *think* they have to be saved to flash for the charge to actually work. This from memory is what the UI does (that 2nd tab that allows you to set # of batts, C rate then say ‘go’).

You can send them to the charger. They can reside in memory (ready to be written to flash), but I don’t *think* (haven’t tested this recently tho) that saying ‘charge’ will work. Umm…. I could be completely wrong.  Hmm. not sure this reply is any use then!

—
Neil Clayton
neil@cloudnine.net.nz

raplin, 2018-05-04:
Yeah, when you start a charge operation you need to specify a memory location.   As an aside, you can adjust the charge current (and voltage) while a charge is running; i.e. you can dynamically adjust charge current based on your own external sensors. (I posted an 'issue' with a few lines of code implementing this)

scornflake, 2018-05-11:
Just to say “I am alive” - just haven’t had time recently to work on the project.

—
Neil Clayton
neil@cloudnine.net.nz

TheBum, 2018-05-12:
No problem. I haven’t been flying except for a few flights at one fun fly, so I haven’t had much use for it.