johncclayton/electric #117: Support for Pi Zero W

sanaraken, 2018-03-28:
I there anyway we can use the Pi Zero W? Not sure how much processing power the Electric need. If the Pi Zero W is enough. Just want a more compact option on smaller cases.

johncclayton, 2018-03-29:
It *should* work as long as there is WiFi and the Hypriot image for the OS
works on the device just as on its larger brother - I've not got one and
thus have never tried it.

Thanks,
--
John Clayton
Skype: johncclayton

sanaraken, 2018-03-29:
I tried and it powers up and able to set up wifi settings, but it will not connect to the icharger. Tried it multiple time and 2 different pi zero w.

sanaraken, 2018-04-04:
I guess theres no hope on the Pi zero w.

closed 2018-04-04 by sanaraken

johncclayton, 2018-04-04:
Why? Seems like it has the right hardware or did you find something out
that causes you to say that it specifically won't work?

Thanks
--
John Clayton
CTO

tsteneker, 2018-04-04:
I kinda think that the Pi Zero W will work. I suspect that the code just checks for a Pi 3. I was going to look into it but haven't had a chance yet. Also not sure if the wifi range on the Pi zero is as strong as the Pi 3B.