johncclayton/electric #118: IPad app not showing both channels and other info

Jaymayville, 2018-04-02:
Hey, I just finished setting everything up in my charge case. On my phone, everything works as it should. On my iPad though it seems to have something going on. It doesn't read the info coming from the charger like Power supply voltage and etc. also, it only shows channel 1 with only one cell. Channel 2 isn't displayed. I have been through all the app settings and such with no luck of fixing this. I've also deleted the app and tried downloading it again but that didn't do anything g either. 

Any ideas? How come everything g works fine on my IPhone but not the iPad?

Jaymayville, 2018-04-02:
I wish I could show you a screenshot of the iPad screen to show you what I mean...

scornflake, 2018-04-02:
You can. Isn’t it one side button plus power button to take a shot? Something like that.

—
Neil Clayton
neil@cloudnine.net.nz

Jaymayville, 2018-04-03:
I know how to take one, I just didn't think I could send a pic on that thread. Here's a couple shots of what I'm seeing on my end. The first pic is of the app on my phone. Everything works fine. The second pic is what shows up on the iPad.

scornflake, 2018-04-03:
Annnndddd. you’re right. they don’t show. I guess you’d need to u/load to dropbox and link from there.
—
Neil Clayton
neil@cloudnine.net.nz

Jaymayville, 2018-04-03:
Ok, thanks. I'll try that out...

scornflake, 2018-11-08:
any luck on a dropbox link?