johncclayton/electric #119: Raspberry Pi 3 Model B+

muzz105, 2018-04-08:
I have been trying to get this working with no joy.
I have tried on both new and old windows computers as well as a Mac, but generally get the same result with just the red light flashing a sequence of flashes. I tried loading Rasbian and saw that there is a green light flashing if it starts loading the software.

The only thing I can think of is that there is a new Raspberry, this is from the site. The Raspberry Pi 3 Model B+ (note the plus on the end) is the latest generation board released by the Raspberry Pi Foundation on 14 March 2018.

Could there be some difference here?

Thanks,

Murray

scornflake, 2018-04-08:
Dunno. Maybe the image is borked somehow. It’s automatically built, we don’t test each one (the process *should* be the same for each).
I’ll have to try it soon and see.

—
Neil Clayton
neil@cloudnine.net.nz

muzz105, 2018-04-09:
Hi, was able to swap my PI with a PI 3 Model B and it worked straight up. So not sure what has changed in the plus version but I know the Model B works as advertised.

Thanks,

Murray

johncclayton, 2018-04-09:
I've ordered a Pi 3 B+ to take it for a spin - maybe I can fix the build
once it arrives (do not hold your breath - my alterego life as a CTO has
stolen all my free time, well, that and the game called world of tanks)

Thanks,
--
John Clayton
Skype: johncclayton

johncclayton, 2018-04-13:
Notes so far: 
- I had to manually restart hostapd then dnsmasq, as the Electric AP didn't come up automatically.  Once restarted it came up OK. 
- get_ip_address.sh code isn't compatible/working.  It looks for "inet addr" on a Pi, but look at the pi3b+ output

wlan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.10.1  netmask 255.255.255.0  broadcast 192.168.10.255
        inet6 fe80::ba27:ebff:fe1f:254c  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:1f:25:4c  txqueuelen 1000  (Ethernet)
        RX packets 197  bytes 15940 (15.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 82  bytes 17456 (17.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Changes required for get_ip_address.sh: 
- look for "inet " (with a trailing space to exclude the inet6)
- change -F seperator arg to " " from ":" in get_ip_address.sh

- Building for the RPI3B+ requires a configuration change in the sd_card image creation - specifically the pirate user DOESNT exist in the image, it's added by cloud-init later on.  Removing this from the sequence of commands works but we're left with a question about how to support different hardware.   SOLUTION: this is likely just an artifact of Hypriot's v1.8.0 image and their use of cloud-init, and very likely doesn't require multiple sd card images because they already mention their single image works for all PI's out there.  

- When I used the Android app's functionality to write the SSID/PSK via the Electric AP (from memory that's a write to the status server URI <ip>:4999/wifi) the changes didn't get stored in /etc/wpa_supplicant/wpa_supplicant.conf, I have no idea why at this point. 

It's 2:47am, I'm off to bed.

johncclayton, 2018-04-13:
I'll continue testing this tomorrow I think.

johncclayton, 2018-04-13:
i mean... later today. lol...  nach muede kommt doof

LiorMetargem, 2018-04-13:
Hello, I would also like to solve the problem. My pi3 B + 2018 also does not work. Only red LED blinks

johncclayton, 2018-04-17:
Correct - the current "master" image will not work on a pi3 b+, it'll blink
all the way til next Christmas and do nothing.

We've got a few hurdles to get over before it does - so this may take a
while (weeks), if you are in a hurry all I can suggest for now it to get a
Pi3 (normal) and go with that - or if you are feeling adventurous dive in a
feel free to help :-) I am working on this under this issue:
https://github.com/johncclayton/electric/issues/119

Thanks,
--
John Clayton
Skype: johncclayton

johncclayton, 2018-04-17:
> Many thanks john. I'll wait for the update. I would gladly help. But I
> have no knowledge of the software

;-)

Thanks
--
John Clayton
CTO

johncclayton, 2018-04-19:
[update]: I've got the system booting up on a 3b+, the challenge now is to work out why the services don't start up in the right order.  Tonight I'm playing with additional options in the /boot/user-data file that should result in a more stable boot sequence / startup.

LiorMetargem, 2018-04-20:
Good news has hope

johncclayton, 2018-04-22:
State of play: 

1. Electric AP doesn't appear on first boot - restarting the hostapd.service solves that, however even with a wlan0 WIFI set up, no routing/bridging appears to be in place - so my phone says "No internet access" via the Electric interface. 

2. I've had to hard-code the country=GB value in wpa_supplicant.conf, apparently WIFI won't even start without this now. 

3. I threw out the Hypriot image entirely and instead went for a bog standard Raspbian Stretch Lite img, added docker + compose and this boots by an order of magnitude faster.  Without learning everything about the new Hypriot + Cloud-init startup sequence (which I have no desire to do); it appears a much better route - however; see 1 and 2 above. 

4. Being just an official Raspbian image + a few extra things added in the sd-card creation process; this boots on 2, 3, 3B and 3B+ 

5. This takes LOTS of time to work through :-)

LiorMetargem, 2018-04-22:
Hey john friend . I did not understand, it works with pi3 B+?

LiorMetargem, 2018-04-24:
Hello john I downloaded the new software for pi3 b+ .
The hiccup lamp flashes and wifi connects. But disconnects after a few seconds. Would appreciate help . What am I not doing well?

johncclayton, 2018-04-25:
The Electric project does not work on a 3B+ at this point.  We are working
on it.

The only resolution at this point is to use different hardware.

Thanks,
--
John Clayton
Skype: johncclayton

LiorMetargem, 2018-04-25:
Thank you very much . I thought it worked 🙃

Baedarlboo, 2018-04-26:
John,
I'm also in to 3D printing and also use pi's for wireless print/monitor 3d printer.  The guys over at Octoprint seems to have figured out the Wifi issues.  Don't know how much this will help but here's a link.

https://github.com/guysoft/OctoPi/issues/505

jvic77, 2018-05-22:
Dose it work now on the pi3b+ or not? 
Just cannot get a pi3b can only see to get a pi3b+ order a pi3b but they have sent me out a pi3b+ just need to know so I can tell them I don’t need pi3b+ I need one I have order.  

Thanks for all the hard work that has been put in to jgetting this to work.

jvic77, 2018-05-23:
In one off John Clayton post says started with a clean install rasp Ian image + a few extra things adds in the sd-card creation process this boots on 2,3, 3b and 3b+. 

Dose this mean if I was to install clean image then follow the instaruction for a clean install from the site I mite be able to get work and my be able to give a hand in helping sort out problems. 

3 off us do know a bit about pi and build a few thing that pi controls so I do understand  basics and may be a bit more.

johncclayton, 2018-05-23:
I'd be happy to have someone help out - right now I've not got a lot of
time to spend on the electric project.  I am quite aware however that if I
make time to help you discover how this thing hangs together, that the time
is well spent / multiples out.

The basic problem is very very simple: services don't seem to start in the
right order when the Pi boots.  The 3b+ does actually work from what I can
tell; but I'm no systemctl / networking expert, nor am I intimately
familiar with the setup of the wlan0/1 interfaces - at least not at the
level where i can see what is going wrong.

I'll give you an example: if you flash the Pi 3b+ with the Raspbian based
image from the 3b+ branch build - it'll not start the dnsmasq services.
Ok.  So when I go into a shell and restart dnsmasq, everything works.
Errr... then I kinda scratch my head and ask: why is that?  This is where I
would love to get some help.

As I said; I'm more than happy to sit together and show you how the system
is "supposed" to work.  If you can then spend some time working out why the
boot sequence trips up on itself that'd be awesome.

Should we get all three of us together on a Skype call, screenshare and
walk through the image creation / flash / debug process?

John Clayton
CTO
E: johnc@filewave.com
A: St. Gallerstr 1, 9500 Wil, St Gallen
<https://maps.google.com/?q=St.%20Gallestr%201%2C%209500%20Wil%2C%20St%20Gallen>
W: https://www.filewave.com

Baedarlboo, 2018-05-23:
Saw a similar question on dnsmasq with Raspbian.  Here's what was suggested.

"If you have "bind interfaces" in your dnsmasq config, try to remove that."

pH7-JP1, 2018-08-05:
I don't know anything about Hypriot, dockers, or any other of the buzz words I have been reading here, but I have started looking into what it takes to do development, just to fix a few bugs that "bug" me. In that process, I saw that on May 27 Hypriot announced version 1.8.0 that supports the B+. Since this announcement post-dates any of this activity, could the solution to this problem be as simple as using this new release as the base for the Electric image?

johncclayton, 2018-08-05:
Possible!! Thanks for the heads up. I have tried a previous image which
also supports the 3b+ the problem though is their startup sequence - or
much more accurately, on first boot the DHCP and DNS interfaces don't
appear online, but restarting the underlying services helps.

This instability means we can't release, and up to this point I've not had
a lot of chance to look into it.

I'd be happy to show anyone the ropes if they have more time though.

Thanks
--
John Clayton
CTO

johncclayton, 2018-08-05:
> I have very little "spare time" over the next 2 weeks and any of that I
> will be reading and trying to get some of the learning curve out of the
> way. That will be followed by 3 weeks of absolutely no spare time and no
> internet access during which I will probably have forgotten 80% of what I
> learned. After that I should actually be able to get serious about looking
> into this. I am using
> https://github.com/johncclayton/electric/blob/master/development/DEVELOPMENT.md
> <http://url> as my starting point. Is this a good place to start or is
> there something even more direct?

Just hit me up when you have a couple of hours and we can rediscover the
project together via screen sharing, thatll speed your journey.

Thanks
--
John Clayton
CTO