johncclayton/electric #124: Patch - adjust charge current while charging (iCharger)

raplin, 2018-05-04:
Hi,
It's possible to adjust the charge current while a program is running on ( iCharger 406 Duo at least).
Due to the way it's done I think you need to (re) set the max cell voltage at the same time (e.g. 4.2v for a lipo)

This is useful to me because I monitor pack temp and specifically per-cell charge current; I have 4S4P packs, I monitor the "P" bridges so I can watch for packs with lower IR getting too much current, yet if the currents are reasonably even and temperatures ok I can increase the charge current substantially.

This works - sorry it's not a PR, should be a trival cut'n'paste:

(add to 'router.py')
`    elif method == "modify_operation":
        return charger.modify_operation(args["channel"], args["current_limit"],args["voltage_limit"])
`
(add to 'zmq_marshall')
`   def modify_operation(self, channel_number, current_limit,voltage_limit):
        return self._send_message_get_response("modify_operation", {
            "channel": channel_number,
            "current_limit":current_limit,
            "voltage_limit":voltage_limit,
        })
`

(add to "comms_layer.py")
`    def modify_operation(self,channel_number,current_limit,voltage_limit):
        channel_number = min(1, max(0, channel_number))
        values_list = (channel_number, VALUE_ORDER_UNLOCK, Order.Modify,current_limit,voltage_limit)

        modbus_response = self.charger.modbus_write_registers(0x8000 + 2, values_list)
        return OperationResponse(modbus_response)
`

Usage e.g.
`
currentLimitBin=int(currentLimit*100)
                voltageLimitBin=int(voltageLimit*1000)
                device_status = self.chargerZMQ.modify_operation(self.channelID, currentLimitBin,voltageLimitBin)
`

scornflake, 2018-11-08:
I've had a quick read. Looks doable, but I need to figure out the UI for it.  At the mo I'm busy getting the next beta out.

scornflake, 2018-11-08:
I know it's been a million years (well, months).
Did you happen to try this out using some REST post calls, by chance?

One of the problems I had with the UI was making successful 'start charging' calls. I'm wondering you have tested this, what experiences you had.

raplin, 2018-11-08:
Not sure about via REST; I use your base icharger/zmq code in a different outer framework (not web based) but 'start charging' zmq function works for me (on i406 duo). My charger ( https://www.youtube.com/watch?v=D5oxO1MAu8o at 2:40 in) is 'full auto' in that it monitors for battery connection and then automatically starts a charge (+adjusts current when running)