johncclayton/electric #127: Creating or editing presets limited to 2 cells

pH7-JP1, 2018-07-21:
Presets created in the charger with more than 2 cells show a blank when editing the preset and only "Auto", 1, and 2 are available in the pulldown. Same when creating a new preset with the app, both Android and iOS.

pH7-JP1, 2018-07-28:
7 days and no comment?? Either no one is watching the issues, or no one believes this issue. In case it is the later, I created a video: [https://youtu.be/fh0Y25Fm1HM](url)

At the beginning of the video you can see that there is no current "Electric Charge Plan" preset. Then I start a charge for 1 5000mah battery at 2C. You can see at the beginning that a 6-cell battery is attached. The error "Number of cells doesn't match the setting" pops up on the charger. You can see why by looking at the Electric Charge Plan that was automatically created - It is for 7.4V (2S).

pH7-JP1, 2018-07-29:
Yep, I understand that. I at least expected that someone else, not necessarily one of the developers, would comment that they had observed the same problem. The video that is part of the readme on the main github page shows that this did work correctly at one time. 

I wrote software myself until I retired 14 years ago. I may look into the problem myself, but for someone who knows their way around this development environment this may be a few minute fix while it could take me many days just to "learn to walk". 

Thanks for your response though. It at least confirms that other people do read the issues here. I wasn't even sure that I posted in the correct place.

scornflake, 2018-11-08:
cells show is fixed in upcoming beta.

closed 2018-11-08 by scornflake