johncclayton/electric #130: Android app lost?

RC-Thoughts, 2018-11-11:
Hi guys!

Cannot find the Android app? iOS app is installed and works but would need the Android for the field-tablet?

Any advice?

johncclayton, 2018-11-11:
IF I had it!  Really sorry - right now I only  just today looked at the dev environment again because I want to ditch the whole docker side of the setup (belief that it introduces too much complexity and makes setup harder.  With a PyPi based install we can hit production + testing setup with a single distribution style).   

I don’t have the latest Ionic 4 stuff on my systems yet – I pinged Neil; he’s working on upgrading all the web UI stuff to Ionic 4, should expect some really nice improvements to the flow of information there.  But he is asleep right now – so I’m gonna have to wait. 

What are you putting together?  I’m curious – want to tell me more?

RC-Thoughts, 2018-11-11:
Too bad, I'll wait for Neil to chime in :)

I'm making a very minimalistic charger-case with 308 Duo and Electric, I'll try to get the RFID to work on that (on both channels, separately) in the future but at the moment goal is to get the case up and running. I have a extra Android tablet I'd like to use for this, would be attached to lid of the case.

iPhone-usage is a working workaround for the time being of course. 

The field-operation part is a bit mystery to me. If I would have the phone connected to Electric Wifi then I'm not able to use internet on the phone. So I made the Electric to connect to phone's hotspot and that works as long as I don't move too far from the case. If I move too far and the hotspot-connection between case and phone breaks electric apparently does not continuously search for it, I need to restart raspi to get it back up to phone hotspot.

Simple solution would be to run "if no network then try to reconnect..:" but not sure if that would break the app's network wizard somehow?

But yeah, will be a nifty little solution, I'll get the whole thing up and running (basically does that already, works great!) and then I'll dig into it and see how it works and go from there :)

I'll promise to post some pics later when hardware is done, so far electronics are on the table, working :)

At this point I'd like to say a big thanks for a truly great work! Installation was so smooth and easy, really fun to play around with!

RC-Thoughts, 2018-11-11:
Did a ugly bash-script, checks for correct SSID, if no found wait's a little to check again. If found re-connects. If already connected wait's a while.

RC-Thoughts, 2018-11-11:
`#!/usr/bin/env bash

# Get values
ssid="xxxxxxx" #Set your SSID here

while :
do
 status=$(iwgetid -r)
 if [ "$ssid" = "$status" ]
 then
  sleep 5
  #echo "Connected"
 else
  while :
  do
   if [ $(iwlist wlan0 scan | grep ESSID | grep $ssid) ]
   then
    #echo "Network found, reconnecting..."
    ifdown wlan0 >/dev/null 2>&1 && ifup wlan0 >/dev/null 2>&1 && sleep 30
    break
   else
    sleep 5
    #echo "Waiting for correct network"
   fi
  done
 fi
done`

And no, Iäm not a coder :)