johncclayton/electric #131: iPhone notifications

RC-Thoughts, 2018-11-11:
- Electric build 694 (downloaded image & dd upgrade, great job on that!)
- iPhone X iOS 12.1.1 Beta

App itself works, command go trough etc but only when connected to home network. Have not been able to test in field-conditions yet.

Trying to use notifications, all permisisons to Electric notifications are ok but if phone is idle and screen is dark no notifications are being shown. When I know operation is finished I can wait for minutes, no notifications. If I open the Electric app the notification is shown. Kinda defeats the purpose? :)

johncclayton, 2018-11-12:
If you want to be on the alpha channel for the android we have created a closed beta group, please send me your email address and I'll add you into the new Ionic v4 testing group.    You can find me on Skype at johncclayton

RC-Thoughts, 2018-11-12:
> If you want to be on the alpha channel for the android we have created a closed beta group, please send me your email address and I'll add you into the new Ionic v4 testing group. You can find me on Skype at johncclayton

Unfortunately my Skype for business did not find you, but email is tero at rc-thoughts.com
Thanks in advance!