johncclayton/electric #133: electric-sd-card_feature_rpi3bx_625.img - WiFi not starting at all

thomasekdahlN, 2018-11-24:
The Wifi network electric is never showing up using 625 image on hardware that worked well with previous version.

thomasekdahlN, 2018-11-24:
[image]

thomasekdahlN, 2018-11-24:
Testet with the 592 image on the same PI - and the Network apperas immideately, and is possible to log on to. This image has problems connecting to my home wifi network - but that is probably another issue.

MahdiB236, 2023-02-15:
> Testet with the 592 image on the same PI - and the WiFi Network apperas immideately, and is possible to log on to. This image has problems connecting to my home wifi network - but that is probably another issue.

Hey Thomas, any chance you still have a working img file? Thanks.