johncclayton/electric #134: Updating failed to manifest not found?

RC-Thoughts, 2019-01-14:
Any ideas?

`HypriotOS/armv7: pirate@iCharger in /opt
$ sudo ./upgrade.sh
sudo: unable to resolve host iCharger
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   775  100   775    0     0   1568      0 --:--:-- --:--:-- --:--:--  1572
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   649  100   649    0     0   2263      0 --:--:-- --:--:-- --:--:--  2269
Latest version is: 880
Deployed is version: 694... going to stop that and use version: 880
docker-ui is running ... stopping it
docker-ui
Removing container docker-ui...
docker-ui
electric-web is running ... stopping it
electric-web
Removing container electric-web...
electric-web
electric-worker is running ... stopping it
electric-worker
Removing container electric-worker...
electric-worker
Running with VERSION_TAG=:880, and execute docker-compose up -d
Pulling worker (johncclayton/electric-pi-worker:880)...
ERROR: manifest for johncclayton/electric-pi-worker:880 not found
`

RC-Thoughts, 2019-05-11:
I hate to ask this but is this project abandoned?

johncclayton, 2019-05-11:
Nope.  We've not cast it away in disgust and horror, yet I'm not 
personally actively working on it - it's in a decent state though.

found? (#134)

scornflake, 2019-05-12:
No, but I’ve had lots of other stuff on. Still
Intend to get to it one day