johncclayton/electric #150: DropBox link with sd card image is not worku

Markus-Rennings, 2020-12-11:


christoferussss, 2021-03-09:
Hi Markus, did you manage to get the image file? I am also trying but no luck so far!
Cheeers
Christo

johncclayton, 2021-03-09:
Ah, 

This is possible - but I would not say simple / easy.  You will need to build the images / runtime yourself as the automated builds to create the ready to flash .img files are long gone (lack of interest in the project). 

The code for the server is OK; I haven't checked the iOS/Android app in a long time - so I'm not sure of that state.  

Let me know your experience level with building / running these things - and we go from there.

johncclayton, 2021-03-09:
Ok, from memory - building the image file is a quite difficult and involved.  My recommendation as the path to follow would be: 

    1. get your raspberry Pi connected to the network - e.g flash an image, boot it, setup wifi etc. 
    2. pull down dependancies, set up a virtualenv and get the system ready to run the code (reference: https://github.com/johncclayton/electric/blob/master/development/install-on-rpi3.sh)

I just looked at this files - and I think another way might be to set up a session together, a video chat, and we walk through it.  I dunno if this stuff runs any more.  Depends on how tech you want to get. 

I'm happy to help someone get this on its feet - but I don't have the time to maintain it right now on my own; thus my offer of help to get it off the ground for you.  

The server side code runs well, the challenge is the actual runtime environment - there is so much to set up (udev + rules, USB hid, startup scripts, WIFI etc - nothing terribly scary or hard; but there are LOTS of moving parts). 

Let me know what you think.  I'm based in Switzerland by the way.

johncclayton, 2021-03-09:
by the way; VERY useful: https://docs.google.com/document/d/12vy4kCue40k26qsqJIa6b5kwuOIhKOWrTJteruaGcJk/edit#heading=h.wg4jkjj04yor

christoferussss, 2021-03-09:
Cool thanks! Will give it a try!

johncclayton, 2021-03-09:
I think you are heading towards a check out & build scenario, the app links and down and the image isnt available.  Sorry its quite a bit harder than just clicking stuff - but this was never an easy install in the first place, and hey, challenges build character right? :-)  (finding the positive in everything).

christoferussss, 2021-03-09:
No problem, wil give it a go. Been in the electronics work as in hardware development and some c firmware stuff.. looked at all your documentation and phone app and have lots of time on hand since I was made redundant recently due to this Corona virus! Live in Cape town South Africa.Sent from my Galaxy

Markus-Rennings, 2021-03-09:
I found an older image in one of my backups the version is 571. Why are they no part of the github repo ? The size is below 2G. I put it on one drive here: https://1drv.ms/u/s!Aifrfu-ze9JAjm8sVG-jt5UU6Pyk?e=kmVina

Markus-Rennings, 2021-03-09:
I can confirm that the IOs app is running on I Phnoe 12 under IOS 14.4.

Markus-Rennings, 2021-03-09:
By the way, thanks John fo this awesom app. I found it very useful. 👍🚁

christoferussss, 2021-03-09:
I did try it on my raspberry pi 4 but no joy.... do not sure if I must get a pi 3?Sent from my Galaxy

johncclayton, 2021-03-09:
wow - thanks for the thanks! i didn't think it was being used. I'll pass it
onto my brother as well (he runs shinywhitebox.com)

yeah, raspberry pi 3 only - i can't recall right now why, there was
something about how the configuration system in a raspberry 4 is different,
for wifi, and we settled on saying "use a 3 at this stage".

background;

I've quit my job as a CTO, going for a sabbatical year.  I'm focusing my
time on learning everything i can about Azure (yesterday was functions and
triggers), C# and so on.  I had briefly considered rewriting the front end
in Xamarin.Forms ... i have no interest in JavaScript nor TypeScript (so
don't ask me to touch those languages).

Markus-Rennings, 2021-03-09:
If you want to learn new stuff and intrested in Xamarin you should have a look on MAUI: https://devblogs.microsoft.com/dotnet/introducing-net-multi-platform-app-ui/

Also there is another cross platform framework called UNO:
https://platform.uno/

MahdiB236, 2023-02-15:
Hey thanks for image file Mark, downloaded it followed the good doc for setup for my 4010duo but no luck on finding the Electric network on my phone once i booted it up:(  Does the project still work for anyone?