johncclayton/electric #176: electric-sd-card.img download?

morfw, 2022-02-03:
Hi

Where can I download the image file for my Raspberry?

Thanks a lot for your anwser.

Best Regards
Walter

johncclayton, 2022-02-05:
Hi 

The project is no longer actively developed nor is the build architecture available.  I’ve left it in place in case someone finds it useful.  

—
John

careyer, 2022-06-01:
Is there at least the last Pi image available somewhere? Would be really appreciated.

MahdiB236, 2023-02-15:
> Is there at least a last good Pi image available somewhere? Would be really appreciated. I only found a very outdated one

Hey careyer, did you ever find one?

careyer, 2023-02-25:
Sadly no. It is a pitty that such a nice piece of software has disappeared from the day of light. :-(